RoboticsBrno/ArduinoLearningKitStarter-library #5: Make universal pin labeling

CornyjK, 2018-05-01:
Label ESP32 and AVR pins so they can be accessed universally (Eg. D1, A1).

CornyjK, 2018-05-01:
Ax are not possible to map on ESP32, because it already defines "ESP32 Dev Module". Possible solution is switching to another board variant eg. "Electronic SweetPeas - ESP320".

yaqwsx, 2018-05-01:
What is the problem exactly?

JarekParal, 2018-05-01:
There is already `#define A1 pinX` probably. Is it true @CornyjK?

And what about create own board configuration for ALKS (for Arduino/PlatformIO)? 

`ESP32 ALKS ...`

CornyjK, 2018-05-01:
@JarekParal 
Adding new board to boards.txt and making it's /variants/ file would work, it would also eliminate requirement of this library (at least pin definitions) for ESP32. Or you meant another way? Maybie they would accept pull request with our board variant.

JarekParal, 2018-05-06:
It would be nice to have own board.txt (or pins_arduino.h) and send pull request to Arduino and PlatformIO.

JarekParal, 2018-06-11:
What we will do with this @CornyjK :-)?

Are the digital pins (D1, D2, ...) on ESP32 implemented?

CornyjK, 2018-06-11:
@JarekParal 
[In my fork](https://github.com/CornyjK/ArduinoLearningKitStarter-library/commit/139e6e0dfbc3dabe9539f40f07ea5a1de380a655)

JarekParal, 2018-06-11:
Ok, could you send pull-request @CornyjK?

CornyjK, 2018-06-11:
@JarekParal 
I think it should be completed first (third point above, also second - send pull request to ESP32 library). In my opinion this should be in next release.

JarekParal, 2018-06-11:
I agree with your answer. Could you define pins in this library for SPI, I2C on the ESP32 as I mentioned in the issue https://github.com/RoboticsBrno/ArduinoLearningKitStarter/issues/21.

CornyjK, 2018-06-11:
[Here](https://gist.github.com/CornyjK/e1ccd2652432f40611512bdc72f1bff3) is preview of file for PR.

JarekParal, 2018-06-18:
You can set the pins for I2C and SPI on ESP32 in this [functions](https://github.com/RoboticsBrno/ArduinoLearningKitStarter-library/commit/16aab29) as `begin()` parameters: 
- https://github.com/espressif/arduino-esp32/blob/master/libraries/Wire/src/Wire.h#L55
- https://github.com/espressif/arduino-esp32/blob/master/libraries/SPI/src/SPI.h#L55

closed 2018-07-19 by CornyjK