RoboticsBrno/ArduinoLearningKitStarter-library #20: Implement tone() HAL

CornyjK, 2019-04-20:
So tone() can run on esp32

CornyjK, 2019-04-20:
[There](https://github.com/lbernstone/Tone) is a library, which wraps `ledc` as a `tone` properly.

bsaid, 2019-04-20:
> [There](https://github.com/lbernstone/Tone) is a library, which wraps `ledc` as a `tone` properly. If my PR gets accepted and library will be added to Arduino index and PIO registry, we will use it as a temporary solution.

Nice, please do any changes after accepting PR #18.

CornyjK, 2019-04-20:
> 
> 
> > [There](https://github.com/lbernstone/Tone) is a library, which wraps `ledc` as a `tone` properly. If my PR gets accepted and library will be added to Arduino index and PIO registry, we will use it as a temporary solution.
> 
> Nice, please do any changes after accepting PR #18.

Those examples will become useless after this issue gets resolved, because a bit customized generic examples will be used.